Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #158: Clarification about the versions

fred-r, 2022-09-16:
Dear all,

I think that today, in Keil uvision, the rule is that only one version of a pack is active at a time.
So, with Keil uvision, it is not possible to have several versions of a same "base component" in a list of choices I guess.
(The version field is "frozen").

But, I can't find it written in the spec.

I think we have only this indication:
```
Tool, software, or silicon vendors may provide a web infra-structure that lists packs that are available from multiple vendors.
Companies like Arm or IAR provide a \ref cp_KeilComPack "Pack Index Service" that uses either a \elem{vendor}.\elem{name}.pdsc or a \elem{vendor}.pidx file to obtain the latest pack versions.
```
So, my point is :
- shall the spec mandate that only one pack version is active at a time ?
- or shall the spec allow the use of several pack versions ?

fred-r, 2022-09-16:
@slhultgren @vgrstm @lmestm @mdortel-stm

fred-r, 2022-09-16:
A side question is also: if we say that only 1 pack version is active at a time, then
- a user creating a new project can select only the latest version of components
- a user opening an existing project based on previous pack versions should be entitled to reuse the project "as is"

Is this consistent ?
If the user reopens the existing project and wants to "complete" it by adding one more component, what happens ?
Is he forced to use a component from new packs versions ?
If so, there is a risk of compatibility break...

Is he forced to upgrade his entire project ?
This means revalidating everything...

So, I believe that we should allow the user to use any pack version installed in his packs store.
And, when opening an existing project based on older pack versions, he should probably have the choice between:
- upgrading the project
- or downloading the older packs

jkrech, 2022-09-16:
@fred-r your observation is incorrect. In MDK/uVision you can select multiple pack versions to be used by a project.
By default uVision uses the latest installed packversion.
Please open the RTE Dialog. At the bottom you find "Resove" "Select Packs" "Details" buttons. Press the "Select Packs" button. Once you uncheck the "Use latest versions of al linstalled Software Packs" you can explicitly configure each installed pack. If you have multiple versions of a pack installed, then you can select one or multiple pack versions.
By default still the latest version of a component is picked, however you can select specific versions of a component to choose a component version from a previous pack version.

mdortel-stm, 2022-09-16:
So far, it is a tool's choice, from what I know.

Considering uVision or STM32CubeMX, which have different ways to bring the user to the selection of component from packs, the consequence is the same : you cannot use components from different version of the same pack.
When you assume that a pack brings only one "software", it seems obvious that taking different components from different versions of the same pack cannot guaranty a functional project.
But if a pack comes with several different software (for example an RTOS and an IP stack), then the restriction we're talking because a problem. Nevertheless, in terms of delivery, it would look weird to deliver uncorrelated items in the same pack. (personal understanding)

So IMHO, yes we could state that all used components of the same pack should be taken from the same version, but allowing to not respect this seems to me to bring more issues than flexibility from my point of view.

It makes me think that pack here has a role of _coherent semantic container_: all components enclosed are dealing with the same _software block_. If we hide packs from the user flow, we somehow loose that notion : if we try to find something to deal with that with the existing specification, we could imagine that we could reuse the bundle as _coherent semantic container_ : all components of the same software block should be placed in the same bundle (just imagination to explain: not a proposal). Or we maybe consider that this container is in fact the pair _class + group_?

In the end, from my perception of the user flow, we are missing that information that a set of components are parts of one software block.

ReinhardKeil, 2022-09-16:
I suggest that we:
  - consider only one pack version in `csolution` - which is different from MDK.
  - consolidate this discussion under #157 as it is highly related.  I mark this issue therefore as "Discussion done" (just to ensure that we need not to read on multiple places.

closed 2022-11-08 by jkrech