Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #166: Handling of config files with multiple toolchains (avoid clashes)

jkrech, 2022-10-18:
There are tool-chain dependent configuration files described in components.
E.g. CMSIS-Core defined `Device:Startup` component containing `startup_<device>.s` assembler file.
Projects supporting multiple toolchains require toolchain specific config file names
`startup_<device>_<toolchain>.s`
Proposal is to enhance PackChk to validate that the use of multiple toolchains do not result in filename clashes.

ReinhardKeil, 2024-01-30:
@jkrech is this still relevant? Can it be closed or transferred into packchk as it relates to this tool?

jkrech, 2024-01-30:
As we are moving from assembler startup files to C startup files one of the primary scenarios for configuration file clashes for tool-chain agnostic projects is gone.

closed 2024-01-30 by jkrech