Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #168: CMSIS-Pack extensions for the current Generator concept

jkrech, 2022-10-25:
In reference to the Generator Proposal here: 
https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md

This change request for the CMSIS-Pack Specification focuses on enhancements to the existing concept:

a) Addition of some pre-defined `category` attribute values for the `<file>` element
  - `genSource` source code templates for the generator
  - `genHeader` header file templates for the generator
  - `genParams` parameter templates for the generator
  - `genInfo` / `genAsset` other information files for the generator
 
b) Clarify in the specification that a single generator can be associated with multiple components. In this case the expectation is that the generator is invoked only once and the generator ensures that all selected components associated with this generator are considered.

c) Clarify that the `$G` variable is only available in `<command>` and `<argument>` element and is substituted by the `<context>.cbuild.yml` file on invocation following the format described here: https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-CBuild-Format.md

jkrech, 2022-10-25:
@fred-r @slhultgren please review this set of high priority updates. What am I missing?

fred-r, 2022-11-18:
LGTM

fred-r, 2023-02-06:
Can it be integrated in the version 2.0 ?

jkrech, 2023-02-07:
Added to the specification.

jkrech, 2023-02-07:
To be implemented in https://github.com/Open-CMSIS-Pack/devtools/issues/722

closed 2023-02-07 by jkrech