Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #169: Add specification of environment variable and default values for location of the CMSIS Pack Root location

jkrech, 2022-10-26:
For tool interoperability we specify 
a) the environment variable `CMSIS_PACK_ROOT` on all host platforms to specify the location where Open-CMSIS-Pack compliant packs are stored on a user's machine.
b) the setting of the above environment is optional in the sense that in case `CMSIS_PACK_ROOT` is not defined on a user's machine the following host specific default directories are used:
```
 | Platform    | Default path              |
 | Linux       | ${HOME}/.cache/arm/packs  |
 | Windows     | ${LOCALAPPDATA}/Arm/Packs |
 | MacOS       | ${HOME}/.cache/arm/packs  |
 | WSL_Windows | ${LOCALAPPDATA}/Arm/Packs |
```

ReinhardKeil, 2024-01-30:
@jkrech I believe this is covered here https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/installation.md#environment-variables.  Suggest to close it.

jkrech, 2024-01-30:
We don't really seem to have a place for "implementation details for interoperability of tools with CMSIS-Pack support" but it is equally important for the end user. Hence I am closing.

closed 2024-01-30 by jkrech