Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #172: Feedback about layers

fred-r, 2022-11-22:
I understand the concept of layers and, I think, the intention behind : deliver preconfigured consistent set of components as a "preconfigured function", like:
- a board
- an applicative layer
- ...to be defined ...

But, I still need to understand the impact on the elements defined here:
https://github.com/Open-CMSIS-Pack/devtools/issues/595

fred-r, 2022-11-22:
And to be more specific : is a library a project or a layer ?
1. if I recompile it 
2. if I deliver only a binary

fred-r, 2022-11-22:
See also: #113 and #80

ReinhardKeil, 2022-12-06:
See here https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Overview.md#software-layers
Is there anything missing?

ReinhardKeil, 2023-01-09:
We will explain how layers can be used in the next TRM meetings.  I believe we can close this issue.