Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #173: Add Hclass and Hgroup predefined values for <part> element classification

silviooliva, 2022-11-23:
Similarly to what is already present in the Open-CMSIS standard for the \<component\> element (see [here](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/element_taxonomy.html#CclassType)) and for improving the \<part\> elements classification, some predefined values for the _Hclass_ and _Hgroup_ attributes of the [\<part\> element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_parts_pg.html#element_part) should be present in the standard.

Like for the _Cclass_ and _Cgroup_ names of the \<component\> element, also the _Hclass_ and _Hgroup_ names can be defined as strings with a minimum length of 3 characters and a maximum length of 32 characters.

In the attached xlsx file I'm proposing some predefined _Hclass_ names and for any _Hclass_ some _Hgroup_ as sub categories.

Adding these predefined values to the standard can help the developers when creating a pack and improve the IDE, where the packs are installed/used, in case there is the need to sort or group the parts by categories.

[Hclass-Hgroup_Predefined_Values.xlsx](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/files/10073946/Hclass-Hgroup_Predefined_Values.xlsx)

silviooliva, 2022-11-23:
Hereafter a short overview on the sections on the Open-CMSIS-Pack standard that will be modified if this proposal will be accepted.

1. Change **Hclass** and **Hgroup** types from _xs: string_ to, respectively, _HclassType_ and _HgroupType_.
[image]

2. Modify description of **/package/taxonomy/description** element since it could be used to describe the part classes along with the component classes.
[image]

3. Add table with **Hclass** and **Hgroup** predefined values.
[image]

silviooliva, 2022-12-02:
As shown in the Example at point 2. of my previous comment, the **/package/taxonomy/description** element should also include among its attributes the **Hclass** and **Hgroup**. 

This should be translated in the PACK.xsd schema through the following **TaxonomyDescritionType** redefinition:

[image]

Along with this redefinition, there is the need to express the following constraint:
the presence of the **Hclass** and **Hgroup** as attributes of a **\<description\>** element excludes the presence of the **Class** and **Cgroup** attributes for the same element (and vice-versa). 
The examples hereafter shows what should be allowed and what not:

[image]

So, to have a robust schema, perhaps this rule (a logical XOR) should be implemented in the PACK.xsd as well.
As far as I know, in XSD v1.1, this can be done using the _xs:assert_ tag, but in XSD v1.0, that is the schema version used by the PACK.xsd, it is not supported.

A workaround, within the XSD 1.0, to solve this issue is to use an _abstract complex type_. The drawback is that a PDSC writer should add some more advanced information in the \<description\> element.

The following picture describes the workaround based on the definition of an  _abstract complex type_ and of two extensions of the abstract type in the PACK.xsd

[image]

The picture below, instead, shows how the not allowed attribute combinations are detected and signaled with a red underline by the IDE.
It shows also the drawback mentioned before: for each **\<description\>** element the type must be specified (using the _xs:type_ tag).

[image]

@jkrech, what do you think? Do you see other ways to solve this issue? 
I'm not an expert of the XSD syntax so maybe I missed something, but I wanted anyway to share these considerations.

jkrech, 2022-12-05:
The main problem I see with your proposal is the compatibility with existing descriptions.
My proposal is to extend the `packchk` utility to validate that C* and H* attributes are not specified concurrently.

silviooliva, 2022-12-05:
Indeed. It was what I wanted to highlight: with this change in the taxonomy, a check on the C* and H* attributes presence is required somewhere. If it is done by the `packchk`, it is ok.
Thank you for your feedback.

jkrech, 2022-12-06:
An alternative approach would be to split the taxonomy for components and parts as I cannot see the real value of sharing it other than that it is a taxonomy as well. In that case we could still make a distinction between `<description>` and e.g. `<Pdescription>` inside `<taxonomy>`.

silviooliva, 2022-12-06:
I'd vote for this approach instead of demanding the check to the `packchk` utility :-)
I'd use `<Cdescription>` for Components and `<Hdescription>` for the Parts (`P` is used for the Processors). Unless you want to maintain `<description>` for Component to preserve the backward compatibility.

jkrech, 2022-12-06:
For consistency we should start with a lower case letter and should append the `P` and for backward compatibility we need to retain `<description>` but could add a component variant in addition.
`<descriptionP>`
`<descriptionC>`

silviooliva, 2022-12-06:
> _For consistency we should start with a lower case letter and should append the `P` (similar to `sourceC`, `sourceAsm`) and for backward compatibility we need to retain `<description>`_

You are meaning that for the `<taxonomy>` we should have three child elements:
`<description>` for backward compatibility (to be used for the Components, but set as deprecated)
`<descriptionC>` for Components
`<descriptionH>` for Parts
Is it correct?

> _... but could add a component variant in addition._

Not sure I've understood. Could you, please, clarify?

silviooliva, 2022-12-06:
> ```
>   <taxonomy>
>      <descriptionP ... >
>      <descriptionC ... >
>      <description ... >
>   </taxonomy>
> ```

Clear. Thanks.

silviooliva, 2023-01-11:
Taking into account the @jkrech's comments, hereafter I'm showing how the `<taxonomy> `element and its child elements would be defined.

[image]

[image]

[image]

jkrech, 2023-01-17:
@silviooliva in our recent discussion about extending the file category types we found that we should consider the ability for "old  tool versions" to be able to correctly process new pack versions. This would mean we should keep `<description>` for component taxonomy and not introduce `<descriptionC>` since this would mean a loss of information in old tools. 
Do you think this approach would be acceptable even though it is not symmetric?

silviooliva, 2023-01-17:
Hi @jkrech,
in the proposal:

- I kept the `<description>` for backward compatibility (tagging it as **DEPRECATED**)
- I introduced `<descriptionC>` for components (recommending to use it instead of the 'old' `<description>`)
- I introduced `<descriptionH>` for parts

In such a way, the "old tool versions" should be able to correctly process new pack versions if `<description>` is used. 

Alternatively, if you think it is safer, I can remove `<descriptionC>` for components and adapt the proposal to the asymmetric approach (keeping `<description>` for components and introducing `<descriptionH>` for parts). To me this solution is acceptable as well.

ReinhardKeil, 2023-01-17:
@silviooliva IMHO you don't need to differ between `<descriptionC>` and `<descriptionH>`.  

Having `HClass` as attribute in `<description>` would assign this description to a parts.  All that is needed is to make `Cclass` optional.

Old tools would not use `<descriptionC>` and therefore **fail to work** with new packs. What needs to be checked is the behavior of old tools without `Cclass` in taxonomy description.   

@edriouk could you investigate that happens without `Cclass` in taxonomy description.

silviooliva, 2023-01-17:
Hi @ReinhardKeil,

I'm not an expert of XSD, but as far as I know with the XSD v1.0, currently used by the standard, it is not possible to manage the constraint we should introduce using only the `<description>`, that is:

_the presence of the **Hclass** and **Hgroup** as attributes of a `<description> `element excludes the presence of the Class and Cgroup attributes for the same element (and vice-versa)._

Please, find my consideration in the comment above: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/173#issuecomment-1335484588

edriouk, 2023-01-17:
Hi,
current implementation requires Cclass attribute. RTE model keeps taxonomy items in a map with taxonomy ID as a key:
`Cclass[.Cgroup[.Csub]]`
We will need to change the key construction to include Hclass and Hgroup attributes.

ReinhardKeil, 2023-01-17:
Another option might be to introduce a new top-level element `<part-taxonomy>`.  IMHO perhaps the cleanest way as it is balanced with <parts>.

silviooliva, 2023-01-17:
> Another option might be to introduce a new top-level element `<part-taxonomy>`. IMHO perhaps the cleanest way as it is balanced with the `<parts>` element.

@ReinhardKeil, your proposal looks good to me as well.

closed 2023-01-31 by jkrech