Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #174: <mountedDevices> element not required for Daughter Boards description

silviooliva, 2022-11-30:
In the Open-CMSIS-Pack-Spec, a development board is described by a quite complete set of information (vendor, name, images, documentation, features, ...). 
This information is provided in the PDSC file through the <board> attributes and child elements.
Among the child elements, both the **mountedDevice** (the microcontroller device mounted on the development board) and the **compatibleDevice** (the compatible microcontroller devices) must have occurrence **1..***, meaning that at least one occurrence of both must be present.

This constraint doesn't allow to describe the development boards that do not have/require a microcontroller mounted on.
This is the case of the extension or expansion or daughter boards, which can be plugged through a connector (ARDUINO UNO, ST morpho, ...) on a mother board, equipped with an MCU, for expanding its features.

To allow the possibility to describe with the Open-CMSIS-Pack-Spec also these kind of boards, the **/package/boards/board/mountedDevice** and **/package/boards/board/compatibleDevice** elements should be optional, that is their occurrence should be set to **0..***.

[image]

As said, the advantage for the users is that they can describe with the standard a larger set of the development boards, in line with what they can already do for the **\<devices\>** and the **\<parts\>**.

In fact, like they can now create CMSIS packs for describing devices (MCUs) and parts (devices other than the MCUs), with this proposal they could create packs for describing both boards mounting MCUs + parts and boards mounting only parts.

jkrech, 2022-12-01:
I can see  where you are coming from, however I think we are weakening the schema if we make the mountedDevice and compatibleDevice optional. How about requiring a pre-defined value "NO_MCU"?

mdortel-stm, 2022-12-01:
@jkrech do you mean that you want to prevent the pack developer from forgetting to think about the mounted device information and possibly miss the addition of the mountedDevice node?

silviooliva, 2022-12-01:
@jkrech, to me your proposal sounds good.
In such a way we should satisfy both the needs of not weakening the schema and of allowing the daughter board description through the standard.

Hence, according to your suggestion, for a daughter board, we should have something like in the example hereafter:

`<mountedDevice Dname="NO_MCU" Dvendor="Generic:5"/>`
`<compatibleDevice Dname="NO_MCU" Dvendor="Generic:5"/>`

Since the _Dvendor_ attribute is required by both the **mountedDevice** and **compatibleDevice** elements, we should use "Generic:5". Otherwise we should add a value like "NO_VENDOR" to the Device Vendors table of predefined value and use it, having something like this:

`<mountedDevice Dname="NO_MCU" Dvendor="NO_VENDOR"/>`
`<compatibleDevice Dname="NO_MCU" Dvendor="NO_VENDOR"/>`

jkrech, 2022-12-01:
> @jkrech do you mean that you want to prevent the pack developer from forgetting to think about the mounted device information and possibly miss the addition of the mountedDevice node?

Correct, we are relying on this information in order to link boards with devices.
If a csolution target-type specifies only a board the tools need to derive the device information.

jkrech, 2023-01-17:
So the summary of the proposal is to add a predefined value for 
`Dname="NO_MCU"`
and
`Dvendor="NO_VENDOR:0"`
to specify a board that does not have a processor device mounted.

silviooliva, 2023-01-31:
PR #199 opened for this issue.

silviooliva, 2023-02-02:
PR #199 merged into the main branch. I think this issue can be closed.

closed 2023-02-02 by jkrech