Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #176: Management of include paths

fred-r, 2022-12-06:
This is a complement to point #175 

In the current spec we can:
- either declare an entire directory as header ==> this directory must be considered when buidling
- or declare a precise list of header files

It seems that in both cases, when it comes to the build system, the CMakeLists.txt is generated with a directive like:
```
set(INC_PATHS
   "c:\myheaders_path"
```

This means that the user code can access any header file from the Inc folder.

This is not only a matter of public and private header files, it is also a matter of making sure we include only what we are supposed to include.

Does it mean that each component must have its own Inc/ folder to keep a proper access control?

fred-r, 2022-12-07:
@massonal

jkrech, 2022-12-07:
From a build perspective there is no difference whether the component specifies an include path or a header file, because the header file needs to be added to the project explicitly via an `include "<header file>"` statement in the source module that requires access to the component. The specified header files of a component act as a contract between the consumer and the provider and the component version must reflect the compatibility of this contract over time. 
Tools should guide the user to locate the header file(s) for a component. In order to keep things simple, a single header file is ideal. E.g. in the uVision editor there is a context menu item (`Insert "#include file"`) that lists the header files of all selected components with the component they belong to. Once the user chooses one, the corresponding include statement is added to the source code.

massonal, 2022-12-07:
Speaking from my past experiments with GCC, I have gotten to know about quite a few flags managing includes, but there are limitations to what's possible (probably by design, so they are unlikely to evolve):
- you can use `-I`/`-isystem`/... to add a headers search directory in scope for the preprocessor;
- you can use `-include` to force-include a header file before the compiler starts on the regular input files;
- BUT you there is no option to put a single header "in scope": it's either a directory or nothing. So you have to group your headers in directories by "recipient", no matter what tools "above" the compiler do.

All this gets me to think, maybe the file categories "include" and "header" should have differing purposes entirely?
Like, "include" would translate to a compiler flag, whereas "header" would just serve to list the files that are part of the component?
And, to go back to #175, "include" could have a public/private attribute to manage access rights -- but not "header", that would not be its point.

jkrech, 2022-12-07:
Let's not break the 30380 component descriptions described in 841 packs and retain the way we describe header files today (a header file with a public include path).

**Proposal**
- for specifying "private" include paths we introduce the additional file categories:
  - `includePrivate`
  - `includeAsmPrivate`
  - `includeCPrivate`
  - `includeCppPrivate`
  - `includeLdPrivate`

  Note: The existing `include*` file categories are always treated as `public` include paths, as already today.

- for (optionally) listing all header files belonging to the component, we add a new file category `headerPrivate`. Adding separate categories per Asm, C, CPP and Linker seems overkill but could be done for consistency.

  Note: `csolution` shall list these files in the *.cbuild.yml but not in the CMakeLists.txt. 

Please provide your feedback. Am I missing any aspects?

massonal, 2022-12-07:
> Let's not break the 30380 component descriptions described in 841 packs and retain the way we describe header files today (a header file with a public include path).

👍
Backward compatibility is essential, of course!

> Proposal

Your proposal is totally in line with what I had in mind; I think it's one of the most straightforward ways to solve the issue at hand.

------------

An other solution I had envisioned is an optional attribute on files, e.g., 
```xml
<file category="header" name="Inc/my_internal.h" visibility="internal"/>
```
And, for compatibility, the default visibility being "public".
(The wording being, of course, flexible: "visibility", or "scope", or ... // "internal", or "private", or ...)

closed 2023-01-20 by jkrech