Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #180: Unclear documentation on what child elements are mandatory to have in case of inheritance

vovamarch, 2022-12-09:
Spec documentation states for inheritable child elements "Occurrence" as "0..*" or "0..1", meaning "optional",  as they can also be defined in other levels. But some of these elements can be actually mandatory to have (even if via inheritance), and errors are returned by PackChk if not found, while other elements with similar "Occurrence" values can be optional to have and do not cause errors. 

For example description of the [<device> element ](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_device) (and its upper layers) gives an impression that only Dname is mandatory in it , but PackChk would report an error if <processor> is not present. However, no errors are reported if other possible elements such as <memory> or <flashinfo> are absent for <device>.

Spec shall provide more clear statement for mandatory and option child elements.

ReinhardKeil, 2025-02-21:
@vovamarch I would argue that devices with only RAM do not require `<flashinfo>`.  We do not force `<memory>` as theoretical it is possible to have devices with no memory at all - just external bus system. 

What triggered this ticket initially?

vovamarch, 2025-02-21:
The ticket is triggered by a draft pack from chip vendor failing to load in uVision. Verification with PackChk also reported an error in the pack pointing to missing  <processor> element in the <device>.  The actual use case was to list <device> as deprecated without any further details, as it was not meant for usage, but rather for information only.

```
        <device Dname="...">
            <deprecated> ..</deprecated>
        </device>
```

Looking only at specification the  <processor> element is not defined as mandatory.  Along with other elements (e.g. <flashinfo>) it is  documented as having "Occurance  0..*", but PackChk seems to have additional logic and treats these elements differently, causing error when <processor> is absent, and no error when <flashinfo> is absent. Reflecting this logic in the spec would be helpful.  Potentially this may apply to other elements as well.

jkrech, 2025-02-21:
Something along the lines of:
`<processor>` and it's attributes are aggregated over `family`, `subfamily`, `device` or if used `variant` sections. If an attribute is inherited from a higher level it is replaced on a lower level.
After aggregation the attributes `Dcore`, `Dversion`, `Dclock`, must be specified.

vovamarch, 2025-02-24:
@jkrech , yes that would clarify the need  for it.