Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #181: Add `side` and `url` attributes to package/boards/board/image element

silviooliva, 2022-12-13:
Currently the standard allows to specify only one image for a board, even if in two formats (large and small).
Sometimes, there could be two needs:

**1.** **To describe a board specifying its top side and bottom side images**
To cover this use case the occurrence of the `/package/boards/board/image` element should be set to `0..2` (instead of `0..1`).
Also, the attribute `side` could be added to the `/package/boards/board/image` attributes, where the `side` value should be either _top_ or _bottom_ (to be declared as predefined values).

**2.** **To have the board picture(s) included in the pack and, moreover, to specify the picture URL to an external website**
To cover this use case, the attribute `url` could be added to the `/package/boards/board/image` attributes. 
The `url` value should specify the link to an external web site hosting, with the same names, the same images in the `small `and `large` attributes. 
The full path to the small and large images is therefore given appending, to the path in `url`, the picture names reported the `small` and `large` attributes, respectively.
This `url` attribute could be used when a pack is not downloaded and locally installed in a tool, but there is the need to show some pack information (in this case, the board pictures) in preview.

Hereafter how the `/package/boards/board/image` element could be modified according to this proposal.

[image]

willlordarm, 2022-12-15:
Hi @silviooliva. It should already be possible to use a url rather than a directory path reference for an image.

> Specifies the path to a small image. The path is relative to the pack base directory or a link to an image on an external web site.

Source: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_boards_pg.html#element_board_image

Adding a `url` attribute might get confusing because of this. It's worth noting that any images in packs (whether they are remotely hosted or local to the pack) end up with a web hosted equivalent on keil.com (e.g. https://keilpack.azureedge.net/content/Keil.B-L475E-IOT01A_BSP.1.0.1/Images/b-l475e-iot01ax-large.jpg) and keil.arm.com (e.g. https://pack-asset-service.keil.arm.com/Keil/B-L475E-IOT01A_BSP/1.0.1/Images/b-l475e-iot01ax-medium.jpg). If you need a hosted version for an online tool you could potentially utilise the APIs behind keil.arm.com to find the hosted images for a given board.

I definitely agree that it should be possible to include more than one image for a board. I think the additional attributes for `side` might be excessive. A more flexible solution might be to remove the limitation on the number of `<image>` elements that can be included in a pdsc (maybe going to `0...5` rather than removing the limit completely). Having `small` and `large` attributes is probably unnecessary since most clients today will happily process images to the correct size for display. If a change were being made here I would recommend removing them so we end up with a list like this:

`<image path="Images/board_image_1.png">`
`<image path="https://images.vendor.com/Images/board_image_2.png"/>`
`<image path="https://images.vendor.com/Images/board_image_3.png"/>`

silviooliva, 2022-12-16:
Hi @willlordarm, 
thanks for your comment.
I knew of the possibility to specify either an url or an internal path through the `small` and `large` attributes. Instead, I didn't know that the images are hosted on ARM/Keil web servers and I was trying to address a similar use case.
So your proposal to utilise the APIs behind keil.arm.com should sound good.

Concerning the removal of the limitation on the number of `<image>` elements and the introduction of `path` attribute, we are aligned.

While, since a Tool can't guess how to qualify an image, I think that it is important to have in the standard an attribute like `side` describing the image view (top, bottom, perspective, ...).
Maybe, considering the `side` attribute as optional, we could also specify the image default view (I think, "top").

`<image path="Images/board_image_1.png" side="top"/>`
`<image path="https://images.vendor.com/Images/board_image_2.png" side="bottom"/>`
`<image path="https://images.vendor.com/Images/board_image_3.png" side="perspective"/>`

silviooliva, 2023-01-09:
Hello @willlordarm, 
any further comment on your side?
How do you think we can proceed with this proposal?
Thanks.

willlordarm, 2023-01-18:
@silviooliva sorry for the delayed response on this.

I think your proposal works well. My only suggestion is that `side` is changed to `view`. I like the idea of including "perspective" as one of the values but it does demonstrate the limitations of the word `side`.

So we have four proposed changes to board image specification:

**1. Add View attribute**

* A new optional attribute `view` is added to `package/boards/board/image` element
* Available image view types are defined in `ImageViewType` as `top`, `bottom` and `perspective`

**2. Add Path attribute**

* Each image element must specify a required `path` attribute that is either an absolute (usually remote) or relative path to the image file.

**3. Deprecate `small` and `large` attributes

* `small` and `large` attributes are deprecated in favour of allowing multiple images.

**4. Remove limit on number of images**

* Limitation on number of image elements that can be appended to the board element is removed.

@jkrech can you confirm you are happy with these proposed changes?

jkrech, 2023-01-18:
Here is an alternative proposal which is extending today's data retaining the approach:
What if we keep the attributes active and simply add `bottom` and `perspective` attributes to the image element allowing to specify all images within one `<image>` element. I do agree that conceptually this is not the best design and if I would start from scratch I would probably use the proposal above.

```
<image large="Images/board_image_1.png"
       bottom="https://images.vendor.com/Images/board_image_2.png"
       perspective="https://images.vendor.com/Images/board_image_3.png"
       small="Images/board_image_1_s.png"
/>
```

I would propose to make `large` required as well as keeping it as the default.
Existing tools will continue to work without needing to deal with multiple `<image>`  tags.

silviooliva, 2023-01-19:
I understand your concern.
I'm ok with this alternative proposal as well. It should anyway fit the need to describe a board with multiple images.
So to me we can proceed with this solution.

jkrech, 2023-01-23:
@willlordarm could you please comment on my proposal?

silviooliva, 2023-01-27:
Hi @jkrech, @willlordarm,
may I proceed with a PR according to Joachim's alternative proposal?

silviooliva, 2023-01-30:
Hi @jkrech,
I opened the PR #198 for this issue.
According to your proposal I set the `large `attribute required, but this is causing some error when running the **[Validate schemata / Validate pack schema](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/actions/runs/4042518217/jobs/6950316041)** check.

[image]

I think we should keep the `large `attribute as optional. Is it ok?

closed 2023-01-30 by jkrech