Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #184: Add clang entry to CompilerEnumType to *.PDSC format

smartinou, 2023-01-03:
In the schema of the .pdsc format (https://raw.githubusercontent.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/v1.7.7/schema/PACK.xsd), add entry "CLANG" as one of the supported TCompiler.

Clang has an ARM backend and should be available for selection.

VGRSTM, 2023-01-03:
@smartinou I'm with you
Will serve a past comment https://github.com/Open-CMSIS-Pack/devtools/issues/220#issuecomment-1075126945

jkrech, 2023-01-17:
Adding `clang` to the `CompilerEnumType` raises the expectation that this compiler is supported end to end. The project has agreed at this point to support a subset of tool-chains: GCC, IAR and Arm Compiler.

ReinhardKeil, 2023-01-17:
Alignment:
  - add `clang` to Open-CMSIS-Pack-Spec so that conditions are extended to support this compiler
  - keep CMSIS-Toolbox support limited to GCC, IAR, AC6 as companies can extended the support of CMSIS-Toolbox with pull requests.

VGRSTM, 2023-01-17:
Adding [Document how a compiler can be added to CMSIS-Toolbox](https://github.com/Open-CMSIS-Pack/devtools/issues/493) as cross reference

smartinou, 2023-01-17:
Great, the current alignment fully satisfies my initial request, which was to add it to the CompilerEnumType of Open-CMSIS-Pack-Spec.

There's already some compilers in the list outside of the ones officially supported by CMSIS-Toolbox. I think it's a good thing to have packchk not choke on such a popular compiler, regardless if the toolbox doesn't support it.

Thanks a lot for following up on this and for the link above.

closed 2023-01-24 by JonatanAntoni