Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #188: How to manage boards with many revision?

jeromecoutant, 2023-01-12:
Hi

https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_boards_pg.html#element_boards

If one board have many revision, but each revision impact only 1 of the feature,
it seems we need to duplicate all the board structure for the new revision (and new uuid).

Maybe we could find a way to avoid this and to create a board level with all common information,
and then a sub level for revision ?

jkrech, 2023-01-17:
Your request reminds me of device variants that we introduced in the specification a long time ago. So the concept by which common information is inherited rather than repeated has been recognized as a means of reducing redundancy. At the same time we find that people generating this information from other sources do not benefit much and it can be harder to read.
I am also worried about existing tools being unable to correctly process this new board description format. I could also imagine that it maybe favorable to release a separate new board support pack for each new board revision assuming that previous revisions won't require any update at all.

jeromecoutant, 2023-01-26:
@avilei

avilei, 2023-01-27:
It's an interesting proposal. Of course we would need to find the right trade-off between reducing redundancy on the one hand, and improving tools efficiency and back compatibility on the other hand.
I'm also adding @silviooliva to the discussion.

jkrech, 2023-03-21:
@avilei and @silviooliva I think we are awaiting a proposal from you.

closed 2023-03-22 by jkrech