Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #190: As a software component developer, I want to keep *.pdsc file in software repository, so I can track it under version control together with code.

DavidJurajdaNXP, 2023-01-17:
There should be way how to use locally cloned repository by standard Open CMSIS Pack tools.

Let's take https://github.com/FreeRTOS/CMSIS-Packs as example (*.pdsc is not part of repositories). 
I think it would be useful to handle pack as image of repository.

ReinhardKeil, 2023-01-17:
Good point, actually there is a way to do it.

cpackget supports local packs could be a repository.  See: https://github.com/Open-CMSIS-Pack/cpackget#adding-packs - `cpackget add path/to/Vendor.PackName.pdsc`

However the AWS CMSIS-Packs have a process to build the *.pdsc file.  Once you have executed that process you could use it as local pack.  

I guess we need to document this in a better way (i.e. tutorial).

JonatanAntoni, 2023-01-17:
Gents, I have to admit I don't understand the use case, yet.

@DavidJurajdaNXP: Could you update the issue description with a full user story, please?

- What is the environment/starting point?
- Which steps does the actor (Software Component Developer) want to execute?
- What does the expected result look like?
- What are the acceptance criteria?

In earlier days we had a issue template for user stories which still can be found [here](https://github.com/ARMmbed/shadowfax-architecture/blob/master/.github/ISSUE_TEMPLATE/user-story.md).

JonatanAntoni, 2023-01-17:
@DavidJurajdaNXP, thanks for clarifying this during today's call. Nevertheless, it might make sense to be more verbose here in case other people reading it.

Basically, I think this use case is already addressed today. Let's check if we need to be more specific about it in the docs.
In csolution one can add any directory as "a pack" which contains a `.pdsc` file. Having the `.pdsc` file of a pack in the root folder of the repository, and given the directory structure of the repo is same to the final pack, one can directly use the repo clone as pack source.

In order to find the repo for a pack archive, we have the [repository tag](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/element_repository.html) in the `.pdsc` file. Tools could use this tag to automatically replace a pack in a project by a clone of the repo.

jkrech, 2023-03-21:
@DavidJurajdaNXP could you please comment, where you still may see gaps?

jkrech, 2023-05-08:
@DavidJurajdaNXP, we can give the recommendation about how a public repository providing a pack should be structured, however in the end it is up for the maintainer to decide the structure.
In some cases maintainers may require the user to run e.g. a script first to consume the repository content as a pack.
Do you see any other options to "push" towards this?

ReinhardKeil, 2024-01-30:
@DavidJurajdaNXP my feeling, this is covered by https://github.com/Open-CMSIS-Pack/SW-Pack-HandsOn?tab=readme-ov-file#pack-development

I understand that the situation with AWS is not ideal, but besides training and explaining best practice, I see no other way.
Note also that the CMSIS_6 software packs are all structured this way (i.e. https://github.com/ARM-software/CMSIS_6/tree/main)

closed 2024-02-06 by ReinhardKeil