Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #194: May we please increase the Cvariant length ?

fred-r, 2023-01-20:
Dear all,

today the Cvariant is specified like this:

<html>
<body>

Cvariant | Defines a variant of a component. Is an optional part of the component ID. The variant specifier is a brief string (for example: release, debug). | xs:string [3..32] | optional
-- | -- | -- | --

</body>
</html>

So we cannot use more than 32 chars for a variant name.

We are currently reaching this limit, wo we would like to propose to extend it to 64 char.

I can handle the PR if this is accepted.

Thanks & Regards,
Fred

ReinhardKeil, 2023-01-20:
Note that each component variant provides also a descriptive text. Keep in mind that users need to type these strings.

Of course we can change the limits, but in an IDE the screen space for such information is typically limited.

fred-r, 2023-01-24:
Hi Reinhard,

probably we misused the Cvariant, I will think further and come back to you.

Would there be a possibility to write a "deny" condition allowing to avoid that a user can select 2 components of the same Cclass (or Cgroup or  CSub) ?

Baiscally, if I could say:

Class = Application
Cgroup / Csub / Cvariant : use it for the unicity of the name
Condition: if this application is selected, do not let the user select another application.

If I deny the Cclass "application", I guess my component will do a "self-reject" ?

Thanks & Regards,
Fred

fred-r, 2023-01-24:
As discussed with @jkrech, renaming the issue

mdortel-stm, 2023-01-24:
I wonder if we may have the need for RTOS as well?
It may be the kind of component which shouldn't present more than once in a project, like the application, which in ST is responsible to provide the `main.c`. Do you agree @fred-r ?

ReinhardKeil, 2023-01-24:
In Open-CMSIS-Pack there is the notion of APIs where a feature like this exists.
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_apis_pg.html#element_api - the attribute `exclusive` makes only one selection possible.  When set to `0` you can select multiple implementations.

Maybe there is a need to extend this further.  Could you share your top-level problem description.   @mdortel-stm refers to an RTOS 'main.c' function. Would this be not a user code template?  An example might be helpful too, so that we can discuss the issue.

mdortel-stm, 2023-01-24:
Looks to me that the API mechanism brings part of a solution, indeed.
Anyway, it works only if the component implements an API.

In case of RTOS, indeed CMSIS-RTOS would allow to prevent only one implementation to be used. But I heard some feedbacks in ST saying that API on RTOS is not that critical, in the sense that user won't change that often their implementation and, more important, abstraction layer impacts performances.

Talking about the example provided by @fred-r, this one is related to `main.c`. Our tool aims at producing, as much as possible, applicative code. The runtime entry point, that I name `main.c` here, must be unique in the project. It can be a code template, it can also be generated code. It does not change the fact that we consider having components bringing an application into the project: in that context, it shouldn't be allow to select two application components because, in my example, they would bring 2 `main.c`.

ReinhardKeil, 2023-01-24:
At `Cclass` level there is the notation of a bundle (https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#Component_Bundle) that perhaps does what you need.

It allows to select a consistent list of components that belong to this `Cclass`.  It is used for example in the CMSIS-FreeRTOS pack:  https://github.com/ARM-software/CMSIS-FreeRTOS/blob/main/ARM.CMSIS-FreeRTOS.pdsc#L891

It allows you to create another RTOS Cclass and by applying bundle you could not intermix components from different bundles.

I get the feeling that we need better training material for pack creation and we should start organizing Cclass names in a better way.

mdortel-stm, 2023-01-25:
That was my comment during last call, indeed: it brings some design constraints as answered @jkrech.
One additional question anyway: what would be the behavior of the system if one pack brings components through a bundle for class RTOS and another pack a component for class RTOS but without bundle?

fred-r, 2023-01-27:
I came to the same idea: using the API.
Let me give a try with this approach for my specific use-case.

mdortel-stm, 2023-01-27:
Anyway, the API does not prevent from two components of class Application to be selected at the same time, as soon as they do not have the same group.

fred-r, 2023-01-27:
Yes, this is also my conclusion after sketching a bit this possibility...

Finally, I think I need the "exclusive" attribute at class level to say: I am the only element of my class allowed in this project.

jkrech, 2023-01-31:
I am not sure what the tools developers will say, but how about adding an attribute to the taxonomy description declaring either all member components of a 
`Cclass`
or
`Cclass:Cgroup`
as exclusive, mimicking  the functionality of the `exclusive` attribute in the `<api>`  element.

```
<taxonomy>
  <description Cclass="Application" doc="https://about/applications">Application components</description>
</taxonomy>
```

The actual use cases for this feature seem sparse to me.
Would this be an "urgent" feature?

fred-r, 2023-01-31:
I still need to investigate the bundles (suggested by @mdortel-stm), but your point @jkrech would be great for me.
To me this is important because if we delibver "Application class" component then we need to make sure that we do not end up with several main() in the project.

fred-r, 2023-02-17:
I gave a try with the bundles and it allows me to enforce my constraint.
We still have the problem pointed by @ReinhardKeil  about the excessive length of the Cbundle name but as it is an xs:string we do not have the 32 char limit.
So, now it becomes more a UI issue probably...

But, at least, I have a solution, so we can take more time to analyze the proposal from @jkrech

ReinhardKeil, 2024-01-30:
@fred-r We try to address this with "taxonomy" in #263.  I would suggest to close this issue.

closed 2024-02-06 by ReinhardKeil