Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #202: Two steps further with the generator

fred-r, 2023-02-06:
Dear all,

some changes have been made with regards to the generator like the introduction of the $G variable.
This variable is important because it allows passing a wider context to the generator.

Now, I am thinking about going 2 steps further (maybe through $G, maybe differently).

**Original consideration**
Each generator is limited to a pack.
So, if we have N packs relying on the same generator, it is duplicated.
==> I would like a way to identify a generator in a unique manner, so that I can refer to it from several packs (no duplicate).

**Added artefact**
As discussed in https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md, new file types have been added.
Some of these files may be shared by several packs.
Today we have to duplicate them.
So, in addition to defining a "unique" generator, it would be good to associate to it some "artefacts" it requires.

**Proposal**
In the continuity of: https://github.com/Open-CMSIS-Pack/devtools/issues/659
We may have a "generator pack" that defines a generator with a unique ID and all the required artefacts.
We may define a new type of pack: "generator" and the packs requiring the generator would have it as their dependency.
The generator, when invoked, would receive:
- the current parameters: DFP pack path...etc...
- the current generator pack path
- the path of the pack(s) hosting the associated component(s)

The drawback is that we start delivering "tools" via a pack but this "generator tool" is not really the tool itself, it is only a bootstrap to this tool.
But, with the taxonomy:
- specific type of pack
- vendor
Each vendor, may ensure to define a "unique" generator.
Or, at least, the generator id should be unique among all "generator packs" included by the pack hosting the component calling the generator.

ReinhardKeil, 2023-02-14:
Our view so far was that a Generator only configures the components for a specific `cClass`, i.e. for CubeMX `cClass=Device`.  This was somewhat implicit (unfortunately never explicitly mentioned).

The idea behind is to support more than just one domain specific generator, i.e.:
  - a separate generator that configure the device properties (pinout, clock system, etc.)
  - a generator that creates a ML Model 
  - a generator that configures middleware parameters (i.e. for a graphic display)

With the new proposed concept the Generator should take control over more than just one `cClass`.  Unfortunately the proposal above seems to consider just one generator and this also reflects addition of   [outdir: - gendir:](https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Input-Format.md#output-dirs)

ReinhardKeil, 2024-01-30:
@fred-r my feeling is that this is outdated. The new [generator integration](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-overview.md#generator-support) works well with CubeMX.  We will show this soon with complex middleware and CubeMX.

ReinhardKeil, 2024-02-06:
@fred-r the [global generator concept](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-operation.md#generator-integration) describes how to register generators. Obviously, these are no longer limited to a pack.

Could you make a assessment if this is still relevant.

fred-r, 2024-02-16:
Hi @ReinhardKeil ,

I would say that the proposal answers the questions I raised in this ticket.

Nevertheless, 2 points are still unclear to me:

> No <generator> element in the *.PDSC file is required when the [Global Generator Registry File](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-operation.md#global-generator-registry-file) is used. The generator="id" attribute of the <component> element in the *.PDSC file is the reference to the - id: list node in the global.generator.yml file.

==> how can we indicate the GPDSC to be generated ? I mean that the GPDS is a kind of overlay for the PDSC invoking the generator. How to make the link ?

> Note:
> 
> The only argument to the run: command is the path to the [Generator Information Index File](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-CBuild-Format.md#generator-information-file). There are no configurable parameters for this utility. The invocation is:
> 
> cbrige <csolution-name>.cbuild-gen-idx.yml
> 

==> how to pass parameters to a generator then ?
Is the `cbuild-gen.yml` file doing this job ?
If so, where is it specified ?

ReinhardKeil, 2024-02-16:
@fred-r The answer is in [cbuild-gen-idx.yml](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/YML-CBuild-Format.md#file-structure-of-cbuild-gen-idxyml). 

This file carries all the information for the generator. ‘output:’ defines the location of ‘cgen.yml’

see here for a demo:
https://github.com/Open-CMSIS-Pack/generator-PoC

It effectively uses this
‘output: …/Examples/Demo1/STM32CubeMX/Target_H7‘

fred-r, 2024-02-19:
Thanks @ReinhardKeil.
And for the gpdsc?
Should we indicate it in the cbuild-gen.yml too ?

jkrech, 2024-02-20:
@fred-r using CMSIS-Toolbox 2.2.1 I am seeing e.g. this output in Blinky+Debug.cbuild-gen.yml

```
  generators:
    - generator: STM32CubeMX
      from-pack: Keil::STM32F4xx_DFP@2.17.1
      path: C:/workspace/stm32F4/STM32F4/RTE/Device
      gpdsc: C:/workspace/stm32F4/STM32F4/RTE/Device/STM32F407IGHx/FrameworkCubeMX.gpdsc
      command:
        win:
          file: C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1/MDK/CubeMX/STM32CubeMxLauncher.exe
          arguments:
            - STM32F407IGHx
            - C:/workspace/stm32F4/STM32F4/Blinky+Debug.cprj
            - C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1
        linux:
          file: C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1/MDK/CubeMX/STM32CubeMxLauncher.exe
          arguments:
            - STM32F407IGHx
            - C:/workspace/stm32F4/STM32F4/Blinky+Debug.cprj
            - C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1
        mac:
          file: C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1/MDK/CubeMX/STM32CubeMxLauncher.exe
          arguments:
            - STM32F407IGHx
            - C:/workspace/stm32F4/STM32F4/Blinky+Debug.cprj
            - C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1
        other:
          file: C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1/MDK/CubeMX/STM32CubeMxLauncher.exe
          arguments:
            - STM32F407IGHx
            - C:/workspace/stm32F4/STM32F4/Blinky+Debug.cprj
            - C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1
```
and the component with the generator reference:
```
    - component: Keil::Device:STM32Cube Framework:STM32CubeMX@1.1.0
      condition: STM32F4
      from-pack: Keil::STM32F4xx_DFP@2.17.1
      selected-by: Keil::Device:STM32Cube Framework:STM32CubeMX
      files:
        - file: C:/Users/foo/AppData/Local/arm/packs/Keil/STM32F4xx_DFP/2.17.1/MDK/CubeMX/run_STM32CubeMX.c
          category: source
      generator:
        id: STM32CubeMX
        from-pack: Keil::STM32F4xx_DFP@2.17.1
```

ReinhardKeil, 2024-04-24:
@fred-r this is quite old - most of it is addressed.  I'm closing it and I suggest you start new issues in case that there are specifics that we still need to work on.  Feel free to re-open in case that you disagree.