Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #205: Moving towards resources management (execution domain)

fred-r, 2023-02-07:
Some time back we strated to discuss the possibility to integrate the rzone concept.
Can we start discussing it ?

Typically, the current rzone format probably needs to be reworked a bit with regards to some items like DnumMpuRegions as the MPU is banked on Cortex-M33.

ReinhardKeil, 2025-02-11:
Covered now by https://open-cmsis-pack.github.io/cmsis-toolbox/Experimental-Features/#resource-management - should be closed once this is complete.

iomint, 2025-02-23:
[After Open-CMSIS-pack weekly call held 25w07, when Open-CMSIS-Pack stakeholders were asked to provide feedback]

Reinhard's proposal in [https://open-cmsis-pack.github.io/cmsis-toolbox/Experimental-Features/#resource-management](https://open-cmsis-pack.github.io/cmsis-toolbox/Experimental-Features/#resource-management) addresses the point raised in this ticket so "technically" the "wontfix" label seems no more true. 
Concerning the content, this new resource management proposal seems interesting, we are going to review and discussed it internally in the coming few weeks.

closed 2025-03-17 by jkrech