Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #209: On ARM Dev Studio 2022.2  -- Debugger FVP fails to connect, with libGL errors  ( gcc 11.3.0 )

rhmcruiser, 2023-02-12:
Environment: ARM Development Studio - Version: 2022.2  - Build: 202220912
Platform: Ubuntu 22.04
gcc version: 11.3.0
kernel:  5.15.0-58-generic

Issue: Debug Connection fails with error: 

 libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /opt/arm/developmentstudio-2022.2/sw/models/bin/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

Background: Created a test RTX project  and the build is successful on ARM DS ( source from examples of ARM DS)

Successful Build: 
***********************************************************************************************
Image component sizes
      Code (inc. data)   RO Data    RW Data    ZI Data      Debug   

      1122          6       2348          4       8572      15918   Object Totals
         0          0         32          0       3584          0   (incl. Generated)
        18          0          0          0          7          0   (incl. Padding)
     13098        474        564        164        100     125443   Library Totals
        28          0          2          0          3          0   (incl. Padding)

=====================

      Code (inc. data)   RO Data    RW Data    ZI Data      Debug   

     14220        480       2912        168       8672     138853   Grand Totals
     14220        480       2912        168       8672     138853   ELF Image Totals
     14220        480       2912        168          0          0   ROM Totals

======================

    Total RO  Size (Code + RO Data)                17132 (  16.73kB)
    Total RW  Size (RW Data + ZI Data)              8840 (   8.63kB)
    Total ROM Size (Code + RO Data + RW Data)      17300 (  16.89kB)
=======================

Finished building target: ../RTX5_Cortex-M33_Blinky.axf

00:08:15 Build Finished. 0 errors, 20 warnings. (took 1s.269ms)

***********************************************************************************************

>>>> PROBLEM statement :  Failure when creating a Debug FVP connection for target 

==> Below is the complete Target Console Log:  ( While starting Debug connection ) 

==== 
Iris server started listening to port 7100
telnetterminal0: Listening for serial connection on port 5000
telnetterminal1: Listening for serial connection on port 5001
telnetterminal2: Listening for serial connection on port 5002
libGL error: MESA-LOADER: failed to open iris: /opt/arm/developmentstudio-2022.2/sw/models/bin/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris: /opt/arm/developmentstudio-2022.2/sw/models/bin/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/iris_dri.so) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /opt/arm/developmentstudio-2022.2/sw/models/bin/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  110
  Current serial number in output stream:  111
===== 

Image location: 
[image]

rhmcruiser, 2023-02-12:
==> A few more details  of the Debug connection of my Cortex-M33 FVP connection
- -
In the Connection tab, platform (Cortex-M33 FVP model) and debug activity ("Bare Metal Debug of Cortex-M33") 
In the Files tab, executable to be downloaded to the target is specified, and Load symbols is ticked to allow debug at source-level.
In the Debugger tab,  control Debug from entry point is selected.
In the OS Awareness tab,  Keil CMSIS-RTOS RTX OS awareness is selected.
-  -

jkrech, 2023-02-13:
@rhmcruiser - thanks for reporting your problem, however I cannot see how this relates to this project repository. Could you please contact the product support team for Arm Development Studio 2022.2
Please visit https://developer.arm.com/ and under "Support" menu on the top right, select "Open a support case".
From the error message it sounds like there is a mismatch of your system and the supported Linux OS versions.
I am closing this issue here.

closed 2023-02-13 by jkrech