Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #226: \<examples\> element: add environment name="cmsis", deprecate \<attributes\>

ReinhardKeil, 2023-04-06:
I suggest for the \<examples\> element:

 - deprecate \<attributes\> as this is to my knowledge nowhere used and we should make it easier to publish examples.
 - use environment name=”cmsis" for projects that are using csolution.yml files.

fred-r, 2023-04-07:
Hi @ReinhardKeil,

on ST side, we plan to use the ```<examples>``` element to deliver our example projects for IDEs like EWARM.
So, we would like to keep all the existing possibilities of the tag.

By the way, how do you plan to make it possible to deliver csolution/cproject-based projects ?

Thanks & Regards,
Fred

ReinhardKeil, 2023-04-07:
The environment name="cmsis" would refer to "*.csolution.yml" based projects with the assumption that an compatible IDE can import them.

Does ST use the \<attributes\> that are currently used for examples?  We are nowhere using it (not on web, not in IDEs), and it just makes it more complicated to publish examples.  Most of the information should be anyway in the *.yml files going forward.

ReinhardKeil, 2023-04-11:
Conclusion on the TRM 11. April 23: make <attributes> optional, but not mandatory.

jkrech, 2023-05-16:
PACK Specification updated to make attributes in example optional.
environment names are not restricted by the specification.

closed 2023-05-16 by jkrech