Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #229: Pre_Include_Global.h Isn't Included for Assembler Files

JonTBeckett, 2023-04-13:
Hi,

I'm working on a project where I need some macros defined for assembly source files as well as C source files. I thought I could do this by adding the macro to Pre_Include_Global_h but I'm seeing now that that doesn't get included for assembly files. 

Is that intended behavior? If so, is there a way to provide macros to all assembly files from a Component?

Thanks for any answer you can provide!

Regards,
Jonathan

jkrech, 2023-04-18:
This needs some investigation.
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#Pre_Include_Global_h

> The build environment shall generate a C/C++ header file with the name Pre_Include_Global.h from all selected components and add this file to the command line as a pre-include file for the build of all modules in the project.

The term "module" is certainly not specific enough to simply exclude assembly files here.
As far as I can see C-Preprocessing of assembler files do not consider Pre_Include_... files.

JonTBeckett, 2023-04-18:
Yeah I suppose I implicitly understood module to mean C/C++ and Assembly files when I read that section of the docs. 

If the intent is for Pre_Include_Global.h to only be used with C/C++ files, would there be any interest in adding better support for Assembly via something similar to Pre_Include_Global.h? A key example of where this could be useful is Azure RTOS ThreadX which checks defines in assembly to configure some of their port-specific code.