Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #230: Add `environment` element at `boards` and `components` level

silviooliva, 2023-04-24:
The `environment` element is used in the standard at [`device`](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_environment) and [`part`](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_parts_pg.html#element_part_environment) level.
It may contains child elements that are specific for a **development tool** (identified by the `environment`'s attribute _name_). The structure of an _element_ inside the `environment` is not specified in the xsd schema file, which gives the **development tool** full control of the _element_ usage.

[image]

So far, the `environment` can be used in elements:
`/package/devices/family`
`/package/devices/family/device`
`/package/devices/family/device/variant`
`/package/devices/family/subfamily`
`/package/devices/family/subfamily/device`
`/package/devices/family/subfamily/device/variant`
`/package/parts/part`

To me, it should be introduced also at `boards` and `components` level since, also for these elements, there could be the need to define some tool-specific information.

So my proposal is to extend the standard introducing the `environment` as (optional) child element of the following elements:
**`/package/components/bundle`**
**`/package/components/bundle/component`**
**`/package/components/component`**
**`/package/boards/board`**

silviooliva, 2023-05-03:
@jkrech, @ReinhardKeil, what do you think? 
The modification should enhance the standard. If you are ok with that, I can take charge of opening the PR.

jkrech, 2023-05-03:
@silviooliva, thanks for the systematic review of the <environment> tag. Adding more places for environment specific information adds flexibility to capture additional tool dependent meta information in the package description.
My worry still is that despite us wanting to specify a standard format we end up with a level of flexibility which will make packs unusable in tools which will ignore this meta information.

silviooliva, 2023-05-03:
@jkrech, I understand your concern, even if the presence of the _mandatory_ elements (and their _mandatory_ attributes) in the standard should provide the set of information ensuring that a pack is working in tools ignoring the meta information.
On the other hand, having such meta information through the `environment` element should favor the increase of the number of tools using the Open-CMSIS packs, so making the standard more and more popular.

jkrech, 2023-05-03:
Understood. @silviooliva  I did not mean to say that you should not prepare the PR adding `environment` more systematically throughout the hierarchy of the description.
I certainly prefer an environment tag being used instead of 'abusing' elements intended for other purposes.

jkrech, 2023-05-08:
Thanks for the PR. We had previously added `<extensions>` to `<component>` upon STMicroelectronic's request.
```
1. June 2022: v1.7.8
   - added extension points description for components as suggested by STMicroelectronics
```
Are we certain we need both?

jkrech, 2025-02-10:
PR got merged.

closed 2025-02-10 by jkrech