Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #240: Would it make sense to have a XOR condition ?

fred-r, 2023-06-28:
Today with the conditions we have 3 possibilities:

<html>
<body>

accept1 | At least one accept must be true to signal a true for the complete condition (OR-Rule) | FilterType | 0..*
-- | -- | -- | --
require1 | All require must be true to signal a true for the complete condition (AND-Rule). | FilterType | 0..*
deny1 | If one deny is true the complete condition becomes false. This element overrules require and accept (AND-NOT-Rule). | FilterType | 0..*

</body>
</html>

ACCEPT is a OR so both options can be selected at the same time.
Would it be useful to have a XOR: N possibilities but only 1 can be selected.

fred-r, 2023-06-28:
@cparata

edriouk, 2023-06-28:
@fred-r, we have discussed that in our team in the past. We found that useful to have a "one of" expression.
However have not decided on how to implement it in a way that supports backward and forward compatibility.
The major problem is adding a new tag. Older tools will simply ignore those elements. 
Adding an attribute to an **accept** expression won't work either, because:
- it would not pass validation if an attribute does not match expression domain
- need to be specified to all accept expressions, otherwise can cause an inconsistency

Another option is to add an "exclusive" attribute to a condition. This will be also ignored by older tools, but I think it could be tolerated.
 
Do you have any other suggestion how to implement it? 

Please also note that currently we have three options for exclusivity that can be used instead of condition: 
1. Cbundle: works on Cclass level
2. Exclsusive API: works on Cgroup level
3. Cvariant: works on Cgroup and Csub levels.

jkrech, 2023-08-22:
@edriouk is this what you have in mind?

```
<condition id="Exclusive RTOS" exclusive="true">
  <accept Cclass="CMSIS" Cgroup="RTOS2" Csub="RTX5" />
  <accept Cclass="CMSIS" Cgroup="RTOS2" Csub="FreeRTOS" />
</condition>
```

If "exclusive" attribute is set, then the body must only contain `<accept>` elements describing component attributes (Cvendor, Cbundle, Cclass, Cgroup, Csub, Cvariant, Cversion).

edriouk, 2023-08-22:
@jkrech , yest that is my suggestion.

mdortel-stm, 2023-08-22:
It looks a nice way to bring the feature.

ReinhardKeil, 2024-01-25:
Is this not duplicating a feature that already exists with `exclusive=1` at <api> level.
https://github.com/ARM-software/CMSIS_6/blob/main/ARM.CMSIS.pdsc#L445

I believe this does effectively the same:

```
<condition id="RTOS2">
  <accept Cclass="CMSIS" Cgroup="RTOS2"  />
</condition>
```

mdortel-stm, 2024-01-30:
@ReinhardKeil the `exclusive` attribute you refer to applies only on APIs. IMO, it does not match the original topic.

iomint, 2025-02-23:
[After Open-CMSIS-pack weekly call held 25w07, when Open-CMSIS-Pack stakeholders were asked to provide feedback]

ST proposes to reconsider this proposal. We believe there will be a real added value to introduce this "one of" feature (or XOR condition) as it will lead to easier and "cleaner" condition description. ST supports the implementation proposal from edriouk and Joachim above:
[image]