Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #246: Adding `image` as child element of the `part` element

silviooliva, 2023-07-31:
Like for the `board` element, also for the `part`, it could be useful having a dedicated child element for referencing at least one `part` image.
Since using the `book` element for that is not recommended, we could reuse the `/package/boards/board/image` definition (reported below) adapting it also for the `part`.

[image]

silviooliva, 2023-08-28:
Hi @jkrech, 
if I understand correctly, you are suggesting to define a new `image` element for the `part` (with `top` attribute instead of `large` and `small`), keeping the `image` for the `board` as it is now. 

For the part the expectation is that, if the `image` child element is present, at least one of its attributes needs to be specified.

Anyway, I'll prepare this PR for your review.

jkrech, 2023-08-28:
@silviooliva, we need to keep the specification of the board unchanged to not break existing descriptions. 
We can have a `clean` specification for the part element, using `top`, `bottom`, `perspective`. I would suggest to make the `top` attribute required, whereas the other two are optional.

silviooliva, 2023-08-29:
@jkrech, PR https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/pull/250 opened for this issue.

closed 2023-09-05 by jkrech