Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #256: Provide a way to access just the `API` from a pack

MiloradCvjetkovic, 2023-09-21:
For building libraries it is usually useful to be able to reference just the API from a pack and not to have a dummy component like `Custom` or similar that is used just to access the API.

ReinhardKeil, 2024-01-29:
This may require an extension to the csolution tool.

The current solution is to provide a "custom" component that provides access to API headers and user code templates for implementation. IMHO this could be also used for the library variant of a middleware component that needs to compile against the API.

Let's discuss how this would be used in a practical implementation.

ReinhardKeil, 2024-02-06:
This affects developers that provide a standardized middleware. Compared to users of this middleware it is only a very small fraction of developers. Making API a selectable option (i.e. in an IDE) it could get confusing to middleware users.

Work-arounds for middleware developers are:
 - use "custom" component (with empty implementation) to access the API.
 - use explicit paths to the API for translating the library, should be not a problem for libraries that expose their own APIs.

ReinhardKeil, 2024-03-01:
Add hints on how to create middleware packs in a documentation

iomint, 2025-02-23:
[After Open-CMSIS-pack weekly call held 25w07, when Open-CMSIS-Pack stakeholders were asked to provide feedback]

ST is aligned with the proposal made during 25w07 Open-CMSIS-Pack weekly call to tag this one "wontfix"

closed 2025-02-25 by jkrech