Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #260: Proposal: Component Taxonomy is insufficient - can Pack Datasheets fill a gap?

jkrech, 2023-10-18:
This issue is to capture feedback to the idea of extracting a human readable document `Pack Data Sheet` from the pdsc.
Today we render searchable device and board pages from pdsc files but not for packs (software).

[PackDataSheet.pdf](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/files/12992664/PackDataSheet.pdf)

ReinhardKeil, 2023-10-18:
[PackDataSheet.pdf](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/files/12992664/PackDataSheet.pdf) highlights **Provided Interfaces** as a weakness.  This is exemplified by lwIP where multiple header files are used to define various interfaces. In addition, the header file name also contains a sub-pathname:

```
#include "lwip/tcp.h"    // defines the tcp API of lwIP
```

iomint, 2025-02-23:
[After Open-CMSIS-pack weekly call held 25w07, when Open-CMSIS-Pack stakeholders were asked to provide feedback]

ST considers the weakness "Provided Interfaces" pointed in this ticket not as a real issue.

jkrech, 2025-02-25:
Note: In the meantime we have introduced Overview.md files, however the content is crafted manually.
The added generation of `cbuild.yml` entries for apis demonstrates the available information that could be automatically generated for each component implementing an API.

```
- api: CMSIS:RTOS2@2.1.3
      from-pack: ARM::CMSIS@5.9.0
      implemented-by: ARM::CMSIS:RTOS2:Keil RTX5&Library@5.5.4
      files:
        - file: ${CMSIS_PACK_ROOT}/ARM/CMSIS/5.9.0/CMSIS/Documentation/RTOS2/html/index.html
          category: doc
          version: 2.1.3
        - file: ${CMSIS_PACK_ROOT}/ARM/CMSIS/5.9.0/CMSIS/RTOS2/Include/cmsis_os2.h
          category: header
          version: 2.1.3
```