Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #261: Self-hosted Pack Index Service

jankap, 2023-10-19:
Hi,
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/createPackPublish.html#cp_KeilComPack mentions that ARM and IAR offers Pack Index Service that can be queried to get cmsis-packs. I think it's like a package registry, is that assumption correct?

How can we create such a service on our own? Ideally using GitLab's (generic?) package registry features. Is there a way?

Thanks.

jkrech, 2023-10-19:
Your assumption is correct, a pack index service does maintain and host a file `index.pidx` list/registry of all publicly hosted packs. The question is what is the motivation for you to create a separate/independent list/registry. If you want to have your packs added to the registry you can provide a url where you host your pidx file and we can add it (see Keil.vidx) below.

Here is the index maintained by Arm: https://www.keil.com/pack/index.pidx
This index is constructed from this vendor index file: 
https://www.keil.com/pack/Keil.vidx
and
https://www.keil.com/pack/Keil.pidx

There is a utility that takes a <vendor>.vidx as input and from that compiles an `index.pidx`. This may not be perfect (Version 0.0.3) but we are happy to take feedback and contributions.

I am not familiar with the GitLab package registry feature, therefore I cannot tell whether there is a way to use it by any means but I am afraid this is not compatible.

jkrech, 2023-10-31:
@jankap , please reopen if you think this issue has not been addressed sufficiently.

closed 2023-10-31 by jkrech