Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #273: Cversion attribute in <condition> element does not allow to change major version

ReinhardKeil, 2024-01-04:
The Cversion attribute in <condition> element specifies a minimum version. The documentation is unspecific on behavior of major version changes. However with [Semantic versioning](https://semver.org/) the change of MAJOR version number implies an incompatible change. The CMSIS tools implement an algorithm that does not allow to cross MAJRO version (except when a explicit version range is specified - todo check).

The documentation needs clarification under: 
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_require

edriouk, 2024-01-05:
I can confirm that packs do not follow the MAJOR rule, they got accepted.
In the current implementation only an explicit range can be used to accept greater version.
for the given range  `Cversion=5.6.0:6` all components from 5.6.0 till 6.0.0 are accepted (6.0.1 gets rejected)

We can still modify most of the tools with: 
- user option (command line, preference setting) to ignore MAJOR  rule
- modify version range comparison to use open expressions like `Cversion="5.6.0:"` to explicitly ignore the MAJOR rule

edriouk, 2024-01-05:
Please note that range `"1.0.0:2.0.0-" `effectively covers all 1.xx.xxx version and excludes any 2.0.0 (even pre-release)
It effectively dose the same job as  `"1.0.0:1.999.999"`  but looks nicer.

jkrech, 2024-01-05:
@edriouk I had not thought about that !!!

ReinhardKeil, 2024-01-08:
@edriouk is it today possible to use `Cversion=5.6.0:7.0.0-` in all our tools? Will this be handled as described by you?  If so, we add it also to the documentation, as this topic is a documentation issue.

jkrech, 2024-01-09:
### Cversion:

Specifies a software component version

**require Cversion:** condition is true if version of component is equal or higher than requested but compatible. According to [Semantic Versioning](https://semver.org/) compatibility requires the **major version** to remain unchanged.

**deny Cversion:** condition is true if version of component is lower than requested.

Version ranges are specified with `min_version:max_version`. The condition is true if the version of the component is equal or higher than `min_version` and lower or equal than `max_version`. If min_version and max_version are equal the version must match exactly. 

Only specifying the min_version `Cversion="4.1.0"` is identical to specifying the version range `Cversion="4.1.0:5.0.0-0"`

edriouk, 2024-01-25:
@ReinhardKeil , semantic versioning works consistently across our tools, including `Cversion=5.6.0:7.0.0-`

closed 2024-01-30 by jkrech