Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #274: Cversion on components within a bundle is not allowed according to documentation

RobertRostohar, 2024-01-12:
Is it allowed for components within a bundle to have different versions (Cversion)?

Would be expected and useful.

It seems to be allowed according to schema and seems to be supported by tools (RTE window in uVision).

But it is not allowed according to documentation:

- https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/cp_PackTutorial.html#cp_Bundles
A bundle is basically a variant on the Cclass level. It specifies the attributes Cclass, **Cversion** and optionally Cgroup and Cvendor … Components within a bundle inherit the attributes set by the bundle and **must not set** these attributes again.
- https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#Component_Bundle
A bundle specifies identical attributes Cclass, **Cversion** and optionally Cgroup and Cvendor for several components. Components within a bundle inherit these attributes set by the bundle and **cannot alter** these attributes.
- https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#element_component
**Cversion**: Defines the version of this component. Is a mandatory part of the component ID. **Must not be specified for a component within a bundle.**

I guess the documentation needs to be updated.

mdortel-stm, 2024-01-12:
You're right, there seems to be an inconsistency.
In the bundle version attribute description, we can read as well:

> Note: components can redefine their version using the Cversion attribute of the component element.

jkrech, 2024-01-12:
How about this:

A component bundle is a named set of components that must share identical attributes for `Cvendor` and `Cclass` by introducing the attribute `Cbundle` which becomes a mandatory part of the component  ID. If `Cvendor` is not specified, the value is inherited from the pack level `vendor` element.

The attributes `Cvendor`, `Cclass`, `Cbundle` as well as `Cversion` are inherited by all component elements within a bundle section. Only the `Cversion` attribute can be redefined by a component element.

Components of the same `Cclass` belonging to a different `Cbundle` shall not be selected concurrently. In this way bundles ensure interoperability across multiple components and restrict the mix and match of components within a `Cclass`.

closed 2024-01-30 by jkrech