Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #275: Limit size of "description" attribute or element

ReinhardKeil, 2024-01-16:
**Problem**

The Open-CMSIS-Pack spec uses the term "brief description", however there is no guidance of a reasonable size or any limit enforcement.

Tools that show such descriptions have however practical limitations (see below)

[image]

**Suggestion**

- Document a recommended size of brief description with 128 (US ASCII) characters, 256 characters max.
- Issue a warning in packchk when recommendation is violated.

FelixWang47831, 2024-01-22:
Hi @ReinhardKeil 
For long description, I find it will  show multiple lines for description information. It holds enough characters and is easy to read. So why not have a multi-line display based on 50 or 60 characters to a line?
For example:
[image]
I think it's more friendly for user to get enough information about the project from Pack Installer. And 256 character limitation could be a hard-to-find limitation for the average developer.

ReinhardKeil, 2024-01-22:
@FelixWang47831 it's about usability. We want to show a "table of content".  There is also a way to link a documentation (could be local or a web URL) that should provide more information.  The example below shows a README.md file that could use the formatting features of Markdown (shown nicely in VS Code).

```
    <example name="Blinky" doc="README.md" folder="Projects/Blinky">
      <description>CMSIS-RTOS2 Blinky example with VIO</description>
      <board name="B-L475E-IOT01A" vendor="STMicroelectronics"/>
```   

In your case the title could be "Shows usage of CMSIS-Driver Ethernet (transmits frames and prints received frames)."

ReinhardKeil, 2024-02-06:
OK to implement the proposed limitations.

jkrech, 2024-02-13:
Raised packchk issue: https://github.com/Open-CMSIS-Pack/devtools/issues/1324

jkrech, 2024-03-12:
Documentation got updated. Packchk update is tracked in above mentioned issue.

https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_package_pg.html#element_package
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_examples_pg.html#element_example