Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #279: Provide full, visible and editable "condition" element directly in component element

tcsunhao, 2024-01-24:
In existing pdsc, the dependencies of components are specified through "condition" mechanism. On component level, just a condition refer id is provided, the detailed definitions of the condition are defined in "conditions" element. If user needs to know details of one component dependency, he needs to search up to down, down to up, many times in the "conditions" to figure out the whole picture which is quite inconvenient. When he/she tries to update it, sometimes it is possible. Here is snap of existing NXP DFP conditions:
[image]

Suggested way to improve:
1. Generally, put full picture of condition into component 
2. component dependencies shall be explicitly specified
3. device, board and other hardware related dependency can still refer condition id.
In this way, users can directly get full picture of the dependencies at the first glance with the most addressed component dependencies and without making the conditions too long and messy with too many hardware data.
Here is one simple example:
[image]

We believe that an human easily readable and editable pdsc files can better populate OCP solution especially in directly github repo daily development usage (not ready packs way)

tcsunhao, 2024-01-25:
The introduction of individual “conditions” element or something like that can be implemented in a backward compatible way.

Here is one NXP manifest dependency example:
[image]

DavidJurajdaNXP, 2024-01-30:
NXP manifest schema: [sdk_manifest_v3.14.txt](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/files/14098082/sdk_manifest_v3.14.txt)
See line 1661: 	<complexType name="componentDependenciesType">

edriouk, 2024-01-30:
Moving component dependencies directly into component description body is relatively easy to implement, however that will break backward and forward compatibility for several tools, in particular :
uVision, CMSIS-Toolbox, packchk: all share the same C++ RTE Model  
Arm DS ,  IAR Embedded Workbench (Java/Eclipse RTE Model)   
Keil MDK (Visual Code)

edriouk, 2024-01-31:
Readability can be improved by applying the following rules when constructing conditions:
1. Use component ID as component condition
2. Separate different types and domains of condition expressions into dedicated sub-conditions
3. Avoid further nested conditions

Here is an example of readable conditions:
```
<component Cclass="Device" Cgroup="SDK Drivers" CSub="common" Cversion="2.4.0" condition="Device.SDK_Drivers.common">
 <!-- component body -->
</component>
<!-- ... -->
<conditions>
  <condition id="Device.SDK_Drivers.common">
    <description> Dedicated condition for Device.SDK Drivers.common component</description>
    <require condition="Device.SDK_Drivers.common_Device"/>
    <require condition="Device.SDK_Drivers.common_Toolchain"/>
    <!-- component dependencies -->
    <require condition="Device.SDK_Drivers.common_Dependencies_Require"/>
    <require condition="Device.SDK_Drivers.common_Dependencies_Accept"/>
    <require condition="Device.SDK_Drivers.common_Dependencies_Deny"/>
  </condition>

  <condition id="Device.SDK_Drivers.common_Device">
    <description> Optional device condition to enable Device.SDK Drivers.common component</description>
    <require Dname="MIMXRT1064*"/>
  </condition>
  <condition id="Device.SDK_Drivers.common_Toolchain">
    <description> Toolchain condition to enable Device.SDK Drivers.common component or its files </description>
    <accept Tcompiler="ARM"/>
    <accept Tcompiler="GCC"/>
  </condition>
  <condition id="Device.SDK_Drivers.common_Dependencies_Require">
    <description> Required dependencies by Device.SDK Drivers.common component </description>
    <require Class="CMSIS" Cgroup="Core"/>
    <require Class="Device" Cgroup="SDK Drivers" Csub="clock"/>
  </condition>
  <condition id="Device.SDK_Drivers.common_Dependencies_accept">
    <description> Accepted dependencies by Device.SDK Drivers.common component </description>
    <accept Class="Device" Cgroup="SDK Utilities" Csub="assert"/>
    <accept Class="Device" Cgroup="SDK Utilities" Csub="assert_lite"/>
  </condition>
  <condition id="Device.SDK_Drivers.common_Dependencies_deny">
    <description> Denied dependencies by Device.SDK Drivers.common component</description>
    <deny Class="CMSIS" Cgroup="SDK Drivers" CSub="another_common"/>
  </condition>
</conditions>
```

tcsunhao, 2024-02-01:
Thanks @edriouk, your suggestions indeed improve the readability to a certain extent.
About op3, honestly, we cannot guarantee about that.

iomint, 2025-02-23:
[After Open-CMSIS-pack weekly call held 25w07, when Open-CMSIS-Pack stakeholders were asked to provide feedback]

From ST perspective, the suggestions from edriouk above are sufficient to address the point, so we are aligned with the "wontfix" proposal.

closed 2025-02-25 by jkrech