Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #297: Allow components that only contain a generator

ReinhardKeil, 2024-03-20:
To start a generator, a component may look like this:

```xml
  <component generator="CubeMX" Cclass="Device" Cgroup="CubeMX" Cversion="0.9.0">
      <description>Configure device or board with STM32CubeMX</description>
  </component>
```
It should be therefore accepted to have a component that contains no files.

jkrech, 2024-03-20:
The downside of lifting this restriction in the schema is that people may add "empty components".
My suggestion previously was to add documentation to the generator component.
```
<files>
  <file category="doc" name="path/to/generator/CubeMX/generator flow documentation" />
</files>
```

ReinhardKeil, 2024-03-20:
Works for me.  I think we can close this issue then.

closed 2024-03-20 by ReinhardKeil