Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #302: Extend Config Wizard with a way to enter symbols or values

ReinhardKeil, 2024-04-18:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/configWizard.html specifies `s` to enter strings. 

There should be also a way to enter symbol or value names (basically the same as strings, but without the surrounding quotes `"`.
Adding `<y>` as another item.

thorstendb-ARM, 2024-04-19:
I'd suggest to allow numbers and values on 'y' tag, but no portions of numbers as they are possible on 'o' tag.
To handle the case if the user does not enter a symbol or value (entry deleted), we should stop scanning for possible values at the line ending (as proposed by @ReinhardKeil ). This breaks with the overal rule in ConfigWizard and will not allow to put the value on the next line, but should work.

Currently supported:
```
//   <y>Value or Define Symbol that specifies number of open files <1-16>
//   <i>Define number of files that can be opened at the same time.
//   <i>Default: 4
#define FAT_MAX_OPEN_FILES      maxFiles

//   <y>Minimum number of open files <1-16>
//   <i>Define number of files that can be opened at the same time.
//   <i>Default: 4
//   <y1>Maximum number of open files <1-16>
//   <i>Define number of files that can be opened at the same time.
//   <i>Default: 4
#define FAT_MAX_OPEN_FILES      (minFiles, maxFiles)

//   <y FAT_MAX_OPEN_FILES>Value or Define Symbol that specifies number of open files <1-16>
//   <i>Define number of files that can be opened at the same time.
//   <i>Default: 4
#define FAT_MAX_OPEN_FILES      maxFiles2
```

Known issues:
- when user deletes the content, no new symbol can be entered (this is because the CW does not now where to insert)
- when user enters a number, a prev. set symbol will be replaced, but then the CW struggles

ReinhardKeil, 2024-04-24:
Thorsten, we just need one symbol per line. Therefore I would use a <newline> as delimiter. 
There is currently no requirement to support a list of identifiers, the 2nd case in your example above needs not to be supported.

closed 2024-04-24 by jkrech

thorstendb-ARM, 2024-10-11:
See PR https://github.com/Arm-Debug/vscode-cmsis-csolution/pull/1839, ConfWiz in CSolution should now support numbers and identifiers on the same object, and the usage of both. This already worked, fixed some minor issues.