Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #311: csolution element: inconsistent documentation

DavidLesnjak, 2024-04-26:
1. [csolution clayer](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_csolution_pg.html#element_clayer):
- It seems that attribute `name` is available in the schema. Is it missing in the specification? 
- Example: Missing attribute `type`, which is required by the specification:
```
<clayer type="Board" path="layers/board/FRDM-MCXA153-MW" file="Board.clayer.yml" copy-to="board/FRDM-MCXA153" condition="FRDM-MCXA153-Board" />
<clayer type="Board" path="board/FRDM-MCXA153" file="Board.clayer.yml" condition="FRDM-MCXA153-Board"/>
```

2. [csolution](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_csolution_pg.html#element_csolution) and [template](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_csolution_pg.html#element_cs_template):
- Typo in example: "copyto" instead of "copy-to" (_Note: in two places_):
```
<template name="Basic csolution project" path="templates/blank" file="Blank.csolution.yml" copy-to="Blank" condition="STM32H7S-Series">
```

jkrech, 2024-04-30:
@ReinhardKeil, please see the feedback on the specification update above.

DavidLesnjak, 2024-05-13:
Sorry, there is no problem with attribute "name". I must have wrongly relied on the shema-checker in notepad++.

closed 2024-05-13 by ReinhardKeil