Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #317: GPDSC Spec clarification: is the GPDSC file used for PDSC elements override

BadreddineBJA, 2024-05-28:
GPDSC concept spec clarification
Assumptions (TBC).
 
- The GPDSC .xsd schema is the same as the PDSC schema, hence any PDSC element declared in the [spec](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/packFormat.html) should be supported by the GPDSC.
- The GPDSC is parsed by csolution and used as a mean to **override** the PDSC definition of components.

Scenario example that we want to achieve:  
Using a packA PDSC defintion like the following:

`<?xml version="1.0" encoding="UTF-8" ?>
<package xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" schemaVersion="1.7.2" xs:noNamespaceSchemaLocation="https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/main/schema/PACK.xsd">
    <vendor>STMicro</vendor>
    <name>packA</name>
    <description>Pack A description</description>
    <generators>
        <generator id="GEN-1">
            <description>GEN1 generator</description>
            <exe>
                <command host="win">gen.bat</command>
                <command host="linux">gen.sh</command>
                
                <argument>GEN-1</argument>
                
                <argument>$G</argument>
                
                <argument>template_model.gpdsc.hbs</argument>
            </exe>
            <gpdsc name="$P/template_model.gpdsc"/>
        </generator>
    </generators>

    <components>
        <component Cclass="Device" Cgroup="generated" Csub="compA" Cversion="0.0.1" generator="GEN-1">
            <description>compA description</description>
            <RTE_Components_h>
                
				#define _RTE_COMPA_FROM_PDSC
            </RTE_Components_h>
            <Pre_Include_Global_h>
                
				#define PRE_INCLUDE_GLOBAL_COMPA_FROM_PDSC
            </Pre_Include_Global_h>
            <Pre_Include_Local_Component_h>
                
				#define PRE_INCLUDE_LOCAL_COMPA_FROM_PDSC
            </Pre_Include_Local_Component_h>
            <files>
                <file category="genAsset" name="templates/compA/stm32_compA_template.c"/>
                <file category="genAsset" name="templates/compA/stm32_compA_template.h"/>
            </files>
        </component>
    </components>
</package>
`

and then generating a GPDSC file with the following content: 

`<?xml version="1.0" encoding="UTF-8" standalone="no"?>

<package xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" schemaVersion="1.0"
    xs:noNamespaceSchemaLocation="https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/main/schema/PACK.xsd">
    <vendor>STMicro</vendor>
    <name>packA</name>
    <description>Pack A description</description>
    <url>path</url>
    <generators>
        <generator id="GEN-1">
            <description>GEN1 generator</description>
            <exe>
                <command host="win">gen.bat</command>
                <command host="linux">gen.sh</command>
                
                <argument>GEN-1</argument>
                
                <argument>$G</argument>
                
                <argument>template_model.gpdsc.hbs</argument>
            </exe>
            <gpdsc name="$P/template_model.gpdsc" />
        </generator>
    </generators>
    {{!-- SELECTED COMPONENTS LIST --}}
    <components>
    	<component Cclass="Device" Cgroup="generated" Csub="compA" Cversion="1.0.0" generator="GEN-1">
            <description>compA description</description>
            <RTE_Components_h>
                
                #define USING_CompA_GPDSC
            </RTE_Components_h>
            <Pre_Include_Global_h>
                
                #define PRE_INCLUDE_GLOBAL_compA_GPDSC
            </Pre_Include_Global_h>
            <Pre_Include_Local_Component_h>
                
                #define PRE_INCLUDE_LOCAL_compA_GPDSC
            </Pre_Include_Local_Component_h>
            <files>
                <file category="source" name="stm32_compA.c" />
                <file category="header" name="stm32_compA.h" />
            </files>
        </component>
    </components>    
</package>
`
The content of the RTE\<context> Generated files is the PDSC list of defines

RTE_Components.h: 
`/*
 * CSOLUTION generated file: DO NOT EDIT!
 * Generated by: csolution version 2.0.0
 * Target:  'debug+main' 
 */
#ifndef RTE_COMPONENTS_H
#define RTE_COMPONENTS_H
/*
 * Define the Device Header File: 
 */
#define CMSIS_device_header "stm32u5xx.h"

**/* STMicro::Device:generated:compA:0.0.1 */
#define _RTE_COMPA_FROM_PDSC**

#endif /* RTE_COMPONENTS_H */`
Pre_Include_Global.h
`
/*
 * CSOLUTION generated file: DO NOT EDIT!
 * Generated by: csolution version 2.0.0
 * Target:  'debug+main' 
 */
#ifndef PRE_INCLUDE_GLOBAL_H
#define PRE_INCLUDE_GLOBAL_H
**/* STMicro::Device:generated:compA:0.0.1 */
#define PRE_INCLUDE_GLOBAL_COMPA_FROM_PDSC**
#endif /* PRE_INCLUDE_GLOBAL_H */
`
Pre_Include_<cclass>_<cgroup>_<csub>.h 

`
/*
 * CSOLUTION generated file: DO NOT EDIT!
 * Generated by: csolution version 2.0.0+p2-g52ab45fa
 *
 * Project: 'app4_SW.debug+main' 
 * Target:  'debug+main' 
 */

#ifndef PRE_INCLUDE_DEVICE_GENERATED_COMPA_H
#define PRE_INCLUDE_DEVICE_GENERATED_COMPA_H

/* STMicro::Device:generated:compA:0.0.1 */
#define PRE_INCLUDE_LOCAL_COMPA_FROM_PDSC

#endif /* PRE_INCLUDE_DEVICE_GENERATED_COMPA_H */
`

What is expected ?
GPDSC component tags are taken into consideration at the conversion step

BadreddineBJA, 2024-05-28:
Tests re-executed using csolution v2.3.0, I get the same result as csolution v2.0.0
 
_RTE_Components.h generated file:_ 
```c
/*
 * CSOLUTION generated file: DO NOT EDIT!
 * Generated by: csolution version 2.3.0
 *
 * Project: 'app6_SW.debug+main' 
 * Target:  'debug+main' 
 */

#ifndef RTE_COMPONENTS_H
#define RTE_COMPONENTS_H

/*
 * Define the Device Header File: 
 */
#define CMSIS_device_header "stm32u5xx.h"

/* STMicro::Device:generated:compA@0.0.1 */
#define _RTE_COMPA_FROM_PDSC

#endif /* RTE_COMPONENTS_H */
```

_**Pre_Include_Global.h**_
```c
/*
 * CSOLUTION generated file: DO NOT EDIT!
 * Generated by: csolution version 2.3.0
 *
 * Project: 'app6_SW.debug+main' 
 * Target:  'debug+main' 
 */

#ifndef PRE_INCLUDE_GLOBAL_H
#define PRE_INCLUDE_GLOBAL_H

/* STMicro::Device:generated:compA@0.0.1 */
#define PRE_INCLUDE_GLOBAL_COMPA_FROM_PDSC

#endif /* PRE_INCLUDE_GLOBAL_H */
```
__**Pre_Include_Local_compId.h**__ 

```c
/*
 * CSOLUTION generated file: DO NOT EDIT!
 * Generated by: csolution version 2.3.0
 *
 * Project: 'app6_SW.debug+main' 
 * Target:  'debug+main' 
 */

#ifndef PRE_INCLUDE_DEVICE_GENERATED_COMPA_H
#define PRE_INCLUDE_DEVICE_GENERATED_COMPA_H

/* STMicro::Device:generated:compA@0.0.1 */
#define PRE_INCLUDE_LOCAL_COMPA_FROM_PDSC

#endif /* PRE_INCLUDE_DEVICE_GENERATED_COMPA_H */
```

jkrech, 2024-06-03:
I reproduced a similar scenario in MDK uVision 5.40 and do see the expected behavior, that the Cversion attribute of the component definition in the GPDSC file needs to be larger or equal to the component version in the PDSC file to override it.
This includes all aspects of the component including  RTE_Components.h, Global and Local Pre-Include definitions.

I converted the uvprojx to csolution and repeated the test and I can see the defines from GPDSC if the component defined in the gpdsc is greater or equal to the Cversion of the component specified in the PDSC file. Otherwise I see the defines coming from the PDSC
I used CMSIS-Toolbox 2.4.0 and 2.3.0 but was unable to reproduce the behavior you are describing above.

BadreddineBJA, 2024-06-04:
Thanks Joachim for the tests, I will re-check the tests following your described scenario about the component version in the GPDSC that should be greater or equal to the PDSC version and confirm the behavior.
i used the csolution "v2.3.0" from the toolbox on "windows" for previous described scenarios,

BadreddineBJA, 2024-06-07:
Hi Joachim,

Thanks for your feedback,
Since test scenarios could differ slightly between what am trying to achieve with the toolbox and the GPDSC vs what you have tested using MDK uvision. Could we schedule a short call to align on the scenario that we are trying to validate and the expected behaviors.

This will be very efficient.
Thanks for sharing your preferred slots.

BR
Badreddine

BadreddineBJA, 2024-06-11:
- Tests re-applied using the toolbox v2.4.0.
- Mocked packs and project example used for test: shared in this private [link](https://github.com/badreddinebenjemaa/GPDSC)
- Test results: 

Following the above recommendations from @jkrech (The Cversion attribute of the component definition in the GPDSC file needs to be larger or equal to the component version in the PDSC file to override it), at the convert step, i can see in the RTE folder that the defines from GPDSC are taken into consideration if the component defined in the gpdsc is greater or equal to the Cversion of the component specified in the PDSC file. Otherwise I see the defines coming from the PDSC. 

I think that this behavior should be documented in the GPDSC spec maybe
for the rest of the applied tests, the GPDSC template manual editing is very error prone since it can cause GPDSC .xsd schema inconsistancies which seems implicitly not detected by csolution (maybe warnings messages to the end user should be added to ease the debug step)

BadreddineBJA, 2024-06-11:
Hi @jkrech, 
 
Thank you for your feedback
Regarding your question, Do you make use of config files in your component descriptions for the generator component and the overwrite in the generated gpdsc?

- No, currently In the applied tests, I am not using the attr=”config” in the gpdsc template.

Following our discussion about the strange behaviors observed and described in this ticket, I re-adjusted the GPDSC template following your recommendation:

-	1. I added the same <generators> node in the GPDSC template.
-	2. I Checked for any open / malformed tag in the GPDSC (applied a packchk on the gpdsc .xsd schema).
-	3. I verified if the GPDSC component used version is greater or equal to the PDSC component version.

After following this scenario:

1.	Compose a project using 2 components: compA, compB .
2.	Generate a GPDSC file with the same PDSC definition of this components (Cclass, Cgroup, Cversion,..) (Same exact version).
3.	Apply a conversion step: ‘csolution convert -s <application>.csolution.yml)

**Observed results:** 
- Am observing the expected behavior (the GPDSC components definitions with their associated tags (RTE_components.h, Pre_Include_Global, Pre_Include_local) are winning (overriding) the PDSC component definitions only if they have the version greater or equal to the PDSC declared one.
**NB: tests where not applied on files coming from the GPDSC**

-	Based on the applied tests, csolution seems to take into consideration the GPDSC file for component override actions.
-	The major challenge that remains and that we are discovering with trials and errors on the GPDSC is that missing attributes in the GPDSC (like the generators tag) or a malformed GPDSC template could cause the tools to crash.

-  Maybe we should enhance the GPDSC Spec by providing a small guideline to describe the list of (PDSC elements) required to exist in the GPDSC template and what is optional:
=> this would confirm our assumption that every PDSC element (described [here](https://open-cmsis-pack.github.io/[Open-CMSIS-Pack](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/index.html)-Spec/main/html/index.html) is supported via the GPDSC.

closed 2026-11-02 by jkrech