Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #334: Introduction of Export Control Classification element in PDSC

OCHSTM, 2024-11-04:
**Problem to solve**
An open-cmsis pack may be bound to export control: if so, the pack should provide the related Export Control Classification Numbers (ECCN) for both United States & European Union

**Proposed solution**
For this, a change request is proposed to open-cmsis consortium to add an optional <export-control-classification> subsection as a child of the <package> element
[image]

E.g., specifying the ECCN numbers for the stm32u5xx_hal_drivers pack would look like this:
`<package …>`<br/>`  <vendor>STMicroelectronics</vendor>`<br/>`  <name>stm32u5xx_hal_drivers</name>`<br/>`  <export-control-classification>`<br/>`    <ECCN-EU>NEC</ECCN-EU>`<br/>`    <ECCN-US>5D992.c</ECCN-US>`<br/>`  </export-control-classification>`

The ECCN codes would be managed through Enums:

- ECCN US codes
[image]

- ECCN EU codes
[image]

Please find attached the proposed change for PACK.xsd 1.7.42:
[PACK_xsd.zip](https://github.com/user-attachments/files/17621088/PACK_xsd.zip)

jkrech, 2024-11-05:
a) it would be good if we could link from the specification to a public document that describes the meaning of the ECC Numbers
b) I would prefer a shorter tag name instead of `<export-control-classification>` e.g. just `ECCN`

OCHSTM, 2024-11-05:
> a) it would be good if we could link from the specification to a public document that describes the meaning of the ECC Numbers 

The official web site for ECCN is the following: https://www.bis.doc.gov/index.php/licensing/commerce-control-list-classification/export-control-classification-number-eccn

Within this web site, the following document lists all the codes with a one sentence definition for each item: https://www.bis.doc.gov/index.php/documents/regulations-docs/13-commerce-control-list-index/file

@jkrech Do you think those 2 links would be enough?

> b) I would prefer a shorter tag name instead of `<export-control-classification>` e.g. just `ECCN`

If the ECCN acronym is sufficiently widespread, why not. The pack developer can still refer to the PACK.xsd grammar comments or to the associated documentation to get the meaning  this acronym...

closed 2024-11-13 by jkrech