Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #335: Allow <environment> extension point at <package> level

OCHSTM, 2024-11-04:
**Problem to solve**
Today, the `<environment name=“tool-x”>` element is used to store information specific to the development tool identified by the name attribute. The content below this element is not constrained and thus `<environment name=“tool-x”>` can be seen as an extension point of the PACK.xsd grammar for the tool-x's specific needs.

In current PACK.xsd version (1.7.42), the `<environment name=“tool-x”>` element can only be a child of the following elements:
[image]

This means there is no way for a pack developer to describe "_tool-x custom information_" that would relate to the whole pack and be applicable to any of the children elements in this pack.

**Proposed solution**
Allow the `<environment name=“tool-x”>` element to be a child of the `<package>` element.

jkrech, 2024-11-05:
Reviewing the other occurrences of `<environment>`, this seems to be the only case where a single occurrence is the maximum.
For consistency, we should add an optional `<environments>` tag with one or more child elements `<environment>`

```
                          <!-- list of environments (elements containing information that is specific for a development tool) -->
                          <xs:element name="environments" minOccurs="0" >
                            <xs:complexType>
                              <xs:sequence>
                                <xs:element name="environment" type="EnvironmentType" minOccurs="1" maxOccurs="unbounded" />
                              </xs:sequence>
                            </xs:complexType>
                          </xs:element>
```

OCHSTM, 2024-11-05:
@jkrech good catch indeed!

closed 2024-11-14 by jkrech