Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #338: debugconfig field missing from PACK.xsd schema

Quaker762, 2024-11-26:
There is a severe mismatch between the Open-CMSIS Pack specification and what's in the `PACK.xsd` file that is causing `packchk` to throw a schema error

I note that the following is in the current revision of the Open-CMSIS Pack spec here: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_device

[image]

While in the `PACK.xsd` schema I see the following:

https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/main/schema/PACK.xsd#L947-L966

The `debugconfig` field is missing. We are currently getting around this by passing `--disable-validation`, however it would be preferable to not have this disabled on the build of our device pack.

Quaker762, 2024-11-26:
Interestingly, this seems to be in the `DevicePropertiesGroup` group, however the following snippet (I have made some redactions of sensitive information) seems to throw an parse error

```xml
      <device Dname="">
        <processor Pname="" Dcore="" DcoreVersion="" Dfpu="" Dmpu="" Ddsp="" Dtz="" Dendian="Little-endian" Dclock=""/>
        <processor Pname="" Dcore="" DcoreVersion="" Dfpu="" Dmpu="" Ddsp="" Dtz="" Dendian="Little-endian" Dclock=""/>

        <compile Pname="" Pdefine=""/>
        <compile Pname="" Pdefine=""/>

        <memory name="FlashUser0" access="rw"  start="0x60000000" size="0x00080000" startup="0" default="0"/>
        <memory name="FlashUser1" access="rw"  start="0x60400000" size="0x00080000" startup="0" default="0"/>
        <memory name="SRAM0"      access="rwx" start="0x20000000" size="0x10000" startup="1" default="1"/>
        <memory name="SRAM1"      access="rwx" start="0x20010000" size="0x10000" startup="0" default="1"/>
        <memory name="SRAM2"      access="rwx" start="0x20020000" size="0xE0000" startup="0" default="0"/>

        <!-- Specify the APs for both core0 and core1. This allows a CMSIS DAP to auto-configure the AP per core -->
        <debug Pname="xyz1" __ap="1" svd="svd_file.svd"/>
        <debug Pname="xyz2" __ap="2" svd="svd_file.svd"/>

        <!-- This line is generating a schema error -->
        <debugconfig default="swd" clock="1000000" sdf="sdf_file.sdf"/>

      </device>
```

jkrech, 2024-11-26:
Please note that the <sequence> in the schema requires to follow the order of `<elements>`.
`<debugconfig>` must be listed before any elements of the DefaultDevicePropertiesGroup:
If you change the order as shown below the schema check should pass.

```
      <device Dname="">
        <processor Pname="" Dcore="Cortex-M33" DcoreVersion="" Dfpu="" Dmpu="" Ddsp="" Dtz="" Dendian="Little-endian" Dclock=""/>
        <processor Pname="" Dcore="" DcoreVersion="" Dfpu="FPU" Dmpu="MPU" Ddsp="DSP" Dtz="" Dendian="Little-endian" Dclock=""/>

        <!-- debugconfig must be before any elements of the DefaultDevicePropertiesGroup -->
        <debugconfig default="swd" clock="1000000" sdf="sdf_file.sdf"/>

        <compile Pname="" Pdefine=""/>
        <compile Pname="" Pdefine=""/>

        <memory name="FlashUser0" access="rw"  start="0x60000000" size="0x00080000" startup="0" default="0"/>
        <memory name="FlashUser1" access="rw"  start="0x60400000" size="0x00080000" startup="0" default="0"/>

        <!-- Specify the APs for both core0 and core1. This allows a CMSIS DAP to auto-configure the AP per core -->
        <debug Pname="xyz1" __ap="1" svd="svd_file.svd"/>
        <debug Pname="xyz2" __ap="2" svd="svd_file.svd"/>

      </device>
```

Quaker762, 2024-11-26:
@jkrech Thanks, that seems to have done it. Please excuse my misreading of the schema.

closed 2024-11-26 by Quaker762