Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #347: Is https://www.open-cmsis-pack.org/ valid site?

escherstair, 2025-03-07:
Not sure if here is the right place to ask for, but searching _cmsis-pack_ with google or bing gives [https://www.open-cmsis-pack.org/](https://www.open-cmsis-pack.org/) as first or second entry.
But the site cannot be accessed anymore because of `ERR_SSL_PROTOCOL_ERROR` neither with Edge, nor with Firefox browser.
Is the above a valid site up to now, or it's an old one (and so it would be better not having it indexed by google or bing)?

Thanks

jkrech, 2025-03-07:
Thanks for flagging this. Yes, the link is a valid site and I not able to reproduce your problem.
` openssl s_client -connect www.open-cmsis-pack.org:443` did not flag anything.
`https://www.ssllabs.com/ssltest/analyze.html?d=www.open%2dcmsis%2dpack.org&latest` does not flag anything.
Do you have any further information that could help to find the root cause that flags the error for you?

escherstair, 2025-03-07:
I've just checked that I can connect from home with the same browser (Firefox 136).
I can't connect from office, so behind corporate firewall (sonicwall) and so on (maybe a proxy? - I can't say now).
I think that something in the corporate network infrastructure can be the issue.

closed 2025-03-09 by jkrech